xarial/xcad #2: Trying to run in .net Core 3.1 cannot find System.Drawing.Common

emersonbottero, 2020-07-31:
XCad.AddIn.RK Automação: Exception: Could not load file or assembly 'System.Drawing.Common, Version=4.0.2.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51'. The system cannot find the file specified.
   at Xarial.XCad.SolidWorks.UI.Commands.SwCommandManager.CreateIcons(CommandGroup cmdGroup, CommandGroupSpec cmdBar, IconsConverter iconsConv)
   at Xarial.XCad.SolidWorks.UI.Commands.SwCommandManager.AddCommandGroupOrContextMenu(CommandGroupSpec cmdBar, Boolean isContextMenu, Nullable`1 contextMenuSelectType)
   at Xarial.XCad.SolidWorks.UI.Commands.SwCommandManager.AddCommandGroup(CommandGroupSpec cmdBar)
   at Xarial.XCad.UI.Commands.IXCommandManagerExtension.AddCommandGroup[TCmdEnum](IXCommandManager cmdMgr)
   at RK.SWAddin.SwAddIn.OnConnect() in C:\Users\emers\Source\Repos\RK.SWAddin\RK.SWAddin\SWAddin.cs:line 41
   at Xarial.XCad.SolidWorks.SwAddInEx.ConnectToSW(Object ThisSW, Int32 cookie)

emersonbottero, 2020-07-31:
I downloaded your example SqlDbEfNetCore and it works fine.

emersonbottero, 2020-07-31:
I changed the Sdk="Microsoft.NET.Sdk.WindowsDesktop" and it Worked. Maybe we should have an Template with all the necessary settings

artem1t, 2020-08-02:
I agree with the template. Currently, it is mentioned in the [docs](https://xcad.xarial.com/extensions/registering#net-core), but only as a recommendation (need to change this as well).

artem1t, 2020-09-13:
Updated documentation to insist of setting this option

closed 2020-09-13 by artem1t