xarial/xcad #5: Updating Combobox based on another comboBox selection change

emersonbottero, 2020-08-01:
this code works great.

` public override IEnumerable<string> ProvideItems(SwApplication app) => Directory.GetDirectories(@"X:/"); `

but how can I show in the next combobox the subfolders of the selected folder in the previous combobox?

[image]

artem1t, 2020-08-02:
I think it needs to be implemented similarly to [control dependencies](https://xcad.xarial.com/property-pages/controls/dependencies/), e.t. ProvideItems should also pass dependent controls and be called every time dependency changed.

artem1t, 2020-09-07:
This is now implemented an available in [preview builds](https://xcad.xarial.com/installation/preview-builds/), will be later merged in release

artem1t, 2020-09-13:
Merged to release

closed 2020-09-13 by artem1t