xarial/xcad #6: Add Suport to Bitmap Button

emersonbottero, 2020-08-02:
[image]

artem1t, 2020-08-02:
I think the best way to implement this is to thread the [IconAttribute] as the icon and generated bitmap button instead of the normal button.

``` cs
[Icon(typeof(Resources), nameof(Resources.btn_icon)]
public Action Button { get; set; }
```

Additionally, add the StandardBitmapButtonIconAttribute to have an ability to assign all standard icons, similar to [StandardControlIconAttribute](https://github.com/xarial/xcad/blob/master/src/Base/UI/PropertyPage/Attributes/StandardControlIconAttribute.cs)

emersonbottero, 2020-08-12:
Hi @artem1t . Do you want me to implement it and send you an PR? 🤓 I may need some instructions but I reeeeeeeealy like the way you code and how you use SOLID principles in the implementation.

artem1t, 2020-08-12:
Hi @emersonbottero, that would be fantastic! I am looking for contributors. Please do not hesitate to reach me on info@xarial.com for any questions. I also recently opened the subreddit: https://reddit.com/r/xcad

artem1t, 2020-09-08:
This is now implemented and available in [preview builds](https://xcad.xarial.com/installation/preview-builds/), will be later merged in release.

- Bitmap button can be added on Action property or on bool property (toggle)
- Image will be automatically altered by the framework so it is not cropped by the button border (as per API docs), so no need to do any modifications with image beforehand
- Both custom and standard images can be used
- API supports both pre 2016 and post 2016 version of SW API

Links:

- [Example](https://github.com/xarial/xcad/blob/dev/docs/_src/PropertyPage/Controls/BitmapButton.cs)
- [Documentation](https://github.com/xarial/xcad/blob/dev/docs/property-pages/controls/bitmap-button/index.md)

emersonbottero, 2020-09-13:
didn't worked for me.
[image]

artem1t, 2020-09-13:
Looks like some issue with image conversion. Could you please attach the image you are using or send it to info@xarial.com?

artem1t, 2020-09-13:
And just to confirm you are using .NET Core 3.1?

emersonbottero, 2020-09-14:
I just sent the e-mail.
I'm using .net framework

artem1t, 2020-09-15:
Thank you. I have confirmed this is related to the image pixel format. If you save it as PNG or 24 bit bmp - issue is not reproducible. 

I have opened bug for that: https://github.com/xarial/xcad/issues/19

closed 2020-09-15 by artem1t