xarial/xcad #9: Set Dimension Exceptions

emersonbottero, 2020-08-05:
I Think mPart.Dimensions["A@Extrude1"]?.SetValue(Dimentions.Asize) should not break the code when there is no "A@Extrude1" in the model, or if it is really necessary we could have a mPart.Dimensions["A@Extrude1"]?.TrySetValue(Dimentions.Asize) that fails silently.

emersonbottero, 2020-08-05:
I did an extension for now, But it is in SwDocument. I just notie I'

``` C#
 public static bool SetDimentionValue(this SwDocument doc, string DimentionName, double Value)
        {
            if (doc == null)
            {
                Debug.Print($"Can't set a dimention of a document with null reference");
                return false;
            }
            else if ((Dimension)(doc.Model).Parameter(DimentionName) != null)
            {
                doc.Dimensions[DimentionName].SetValue(Value);
                return true;
            }

            Debug.Print($"Dimention {DimentionName} not found in Model {doc.Title}");
            return false;
        }
```

artem1t, 2020-08-06:
I like the idea with TrySetValue/TryGetValue, trying to make it more consistent with IDictionary, i.e. exception if key does not exist (i.e. dimension is not present)

`bool TrySetValue(string dimName, double val)`

We can add it to the extension method of [IXDimensionRepository](https://github.com/xarial/xcad/blob/master/src/Base/Annotations/IXDimensionRepository.cs). It is similar to [IXPropertyRepositoryExtension::Set](https://github.com/xarial/xcad/blob/master/src/Base/Data/IXPropertyRepositoryExtension.cs)

artem1t, 2020-09-09:
Implemented in the preview builds to be merged into release

artem1t, 2020-09-13:
Merged to release:

IXDimensionRepository.TrySetDimensionValue
IXDimensionRepository.TryGetDimensionValue

closed 2020-09-13 by artem1t