xarial/xcad #20: BitmapButton bool not firing propertyManagerPage DataChanged Event

emersonbottero, 2020-09-18:
``` C#
        [Description("Barra Quadrada")]
        [BitmapButton(typeof(Resources), nameof(Resources.Barra_Quadrada), 48, 48)]
        [ControlOptions(left: 37, top: 0)]
        public bool SquareBar { get; set; } = false; 
```
``` C#
        private void M_Page_DataChanged()
        {
            var test = m_Data;
            
                //TODO: handle the data changing, e.g. update preview
        }
````

if I set a breakpoint on var test = ... and click on the button the breakpoint is never reached
On combobox change the breakpoint is reached

closed 2020-09-23 by artem1t

emersonbottero, 2020-09-24:
is it available on the last dev Nugget Package?

artem1t, 2020-09-25:
That is already in master and published to the public 0.6.1 nuget package

commit 3abc9038 (2020-09-23): Merge branch 'docs' into dev

Fixed #20 - BitmapButton bool not firing propertyManagerPage DataChanged Event
Implemented #21 - Add IXApplication::WindowHandle
Implemented #22 - Add SwApplication::GetInstalledVersion static method