xarial/xcad #45: xcad does not compile (.NET5.0) due to RegAsm.exe

dlucik, 2020-11-12:
dotnet.exe build:

```
RegAsm : error RA0000: Could not load file or assembly 'Xarial.XCad.SolidWorks, Version=0.6.8.0, 
Culture=neutral, PublicKeyToken=60dcaf351d4060db' or one of its dependencies. The system cannot find the file specified. [C:\Users\lukasz.dlucik\Desktop\WSPOLNY\solid50\solid50.csproj]
C:\Users\lukasz.dlucik\.nuget\packages\xarial.xcad.solidworks\0.6.8\build\Xarial.XCad.SolidWorks.targets(8,4): error MSB3073: The command "%windir%\Microsoft.NET\Framework64\v4.0.30319\RegAsm.exe "C:\Users\lukasz.dlucik\Desktop\WSPOLNY\solid50\bin\Debug\net5.0\solid50.dll" /codebase" exited with code 100. [C:\Users\lukasz.dlucik\Desktop\WSPOLNY\solid50\solid50.csproj]
```

my .csproj 
```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFramework>net5.0</TargetFramework>
    <EnableComHosting>true</EnableComHosting>
<AutoGenerateBindingRedirects>true</AutoGenerateBindingRedirects>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="xarial.xcad.solidworks" Version="0.6.8" />
  </ItemGroup>

</Project>
```

artem1t, 2020-11-12:
I have not yet ported this to NET 5. It is not using registry free COM, but RegAsm should not be called. RegSvr32 is used for NET Core and .NET 5. I believe the issue is [targets](https://github.com/xarial/xcad/blob/master/build/Xarial.XCad.SolidWorks.targets) file which is missing the case for net5 and fallback on NetFramework while should fallback on NetCore.

artem1t, 2020-11-12:
As a quick workaround, you can disable the automatic registration with <XCadRegDll>false</XCadRegDll> property and manually call regsvr32, but at this stage, I would recommend writing SW add-ins in .NET Framework until SOLIDWORKS officially supports .NET5/.NET Core, there were few issues reported in .NET Core when using EF, there might be more which we are not aware of. I you are writing stand-alone then it is OK to use .NET Core/NET5.

dlucik, 2020-11-13:
I figure out how to manually registry COM .dll  

```
dotnet build
dotnet publish -r win-x64 -c Debug
regsvr32.exe  .\bin\Debug\net5.0-windows\solid.abastran.tools.comhost.dll
```