xarial/xcad #76: Docmanager GetDocumentType for .sldprt .sldasm and .slddrw gives notsupported

EddyAlleman, 2021-06-12:
should be:

 internal abstract class SwDmDocument : SwDmObject, ISwDmDocument
    {
        internal static SwDmDocumentType GetDocumentType(string path)
        {
            SwDmDocumentType docType;

            if (!string.IsNullOrEmpty(path))
            {
                switch (System.IO.Path.GetExtension(path).ToLower())
                {
                    case ".sldprt":
                        docType = SwDmDocumentType.swDmDocumentPart;
                        break;
                    case ".sldblk":
                    case ".prtdot":
                    case ".sldlfp":   
                    case ".sldasm":
                        docType = SwDmDocumentType.swDmDocumentAssembly;
                        break;
                    case ".asmdot":   
                    case ".slddrw":
                        docType = SwDmDocumentType.swDmDocumentDrawing;
                        break;
                    case ".drwdot":

                    default:
                        throw new NotSupportedException("Only native SOLIDWORKS files can be opened");
                }
            }

EddyAlleman, 2021-06-12:
Just checked the code on github and it is ok.
Something must have messed up my fork (That I made today)

Sorry for the fuzz

EddyAlleman, 2021-06-12:
[image]

artem1t, 2021-07-02:
@EddyAlleman, this was fixed in the current version (before this issue is opened). The code above should pick .sldprt as part. Please make sure that you targeting the latest version from the nuget (this was the issue in the older version).

closed 2021-07-02 by artem1t

EddyAlleman, 2021-07-03:
Hi Artem,

I must be missing something.
The following file still has those errors:
[](https://github.com/xarial/xcad/blob/master/src/SwDocumentManager/Documents/SwDmDocument.cs)
https://github.com/xarial/xcad/blob/master/src/SwDocumentManager/Documents/SwDmDocument.cs

Are you sure you did update your local code to github?

In my fork, there are no Xarial nuget pacakges that need updating.

Cheers,

Eddy