xarial/xcad #87: What is lblMessage.Text in the CustomWinControl example?

flycast, 2021-11-16:
Not a bug but a question.

Following [https://xcad.xarial.com/property-pages/controls/custom/](https://xcad.xarial.com/property-pages/controls/custom/)

What does `lblMessage.Text` represent/come from?

```
using System;
using System.Windows.Forms;
using Xarial.XCad.UI.PropertyPage;

namespace Xarial.XCad.Documentation.PropertyPage.Controls
{
    public partial class CustomWinFormsControl : UserControl, IXCustomControl
    {
        public event CustomControlValueChangedDelegate ValueChanged;

        private CustomControlWinFormsModel m_Model;

        public CustomWinFormsControl()
        {
            InitializeComponent();
        }

        public object Value 
        {
            get
            {
                if (m_Model != null)
                {
                    m_Model.Text = lblMessage.Text;
                }
                return m_Model;
            }
            set
            {
                m_Model = (CustomControlWinFormsModel)value;
                lblMessage.Text = m_Model.Text;
                ValueChanged?.Invoke(this, m_Model);
            }
        }
    }
}
```

artem1t, 2021-11-16:
This is just a Windows Forms Label control text. The idea of this example is to bind the text value from the data model of the property manager page to the custom control with label (textblock) and this label value is dynamically bound to the datamodel property.

flycast, 2021-11-29:
@artem1t : Sometimes I get stuck on the simplest things!

I have everything created as your examples show. My user control is showing up:

[image]

I don't seem to be able to figure out how to set the value of the label `lblText.Text`.

I create the tab variable in SwAddInEx:

```
public ISwPropertyManagerPage<WinFormsPMPage> m_WinFormsPMPage;
m_WinFormsPMPage = CreatePage<WinFormsPMPage>();
```

 and then create the tab itself in my documentHandler OnInit() when the doc loads (I know, this needs work because it creates duplicates everytime I load a file):

` _addIn.CreateTaskPaneWinForm<WinFormsUserControl>();`

Then I am stuck. How do I get and set properties on my custom user control?

artem1t, 2021-12-01:
@flycast , CreateTaskPaneWinForm returns the IXPanel which has a property called Control. In your case this will return the pointer to WinFormsUserControl so you can expose methods or properties to access the controls.

flycast, 2021-12-02:
@artem1t - Thank you!

closed 2021-12-02 by flycast