xarial/xcad #88: the .net control will be lost when show a new PMPpage(same as do again this function) .

painezeng, 2022-01-17:
I want show again when use PMP page onclosed. but custom control will be lost.

[image]

[image]

artem1t, 2022-01-17:
Is this a WinForms or WPF control? Could you please also clarify the version of xCAD you are using? This was the issue in older version which was fixed in some of the previous versions of xCAD

painezeng, 2022-01-17:
it's a winforms control. xcad version is 0.7.11.

artem1t, 2022-01-17:
Thank you, I will double-check, could you please also clarify what do you target (is this .NET Framework or .NET Core?)

painezeng, 2022-01-17:
.NET Framework . i will use latest version to test again.    for me i want use pmp show some ui, then click action button （not the top button "OK"）  to create sketch. but now the pmp page will be close automatic . so i want show the pmp again,also can keep lasted enter information.   maybe you have some good idea to do this function? thanks you. xcad is a good product. i study so many on your websit.

painezeng, 2022-01-17:
i tested with Xcad 0.8.0.0. this issues still exist.