xarial/xcad #89: Changing title of Command Manager Enum Creates New Tab

geoffgscott, 2022-05-29:
Whenever the title of the command group enum is changed I am seeing a new tab created and the old one remains. 

I am also having issues with CommandItemInfo attribute causing the button to be hidden in all documents types.

I am wondering if both are related to me not resetting something correctly when modifying the command group Enum and restarting the process?

XCad 7.12.0 and Solidworks 2021. 

```C#
        [Title("A Title")]
        public enum Commands_e
        {
            [Title("Setup Sync")]
            [Description("Configure sync parameters")]
            [CommandItemInfo(WorkspaceTypes_e.Part)]
            Setup,
            [Title("Sync All")]
            [Description("Syncronize all parameters and set varables.")]
            [CommandItemInfo(WorkspaceTypes_e.Part)]
            PullAll,
            [Title("Sync Changes")]
            [Description("Update changed values")]
            [CommandItemInfo(WorkspaceTypes_e.Part)]
            PullChanges,
            [Title("Get Props")]
            [Description("Get Properties")]
            [CommandItemInfo(WorkspaceTypes_e.Part)]
            PullProps
        }
```

weianweigan, 2022-12-23:
I've had this problem before.It caused by solidworks cache, so there is a old one.

* Open your regedit with path below.
```path
HKEY_CURRENT_USER\SOFTWARE\SolidWorks\SOLIDWORKS 2022\User Interface\CommandManager
```
[image]

* Delete the context that you created.

* Restart SolidWorks and It may works.

andrMollo, 2023-10-14:
HI!
I'm sorry to resurrect this issue but I'm facing the same problem on version 0.7.12 with both SW 2022 SP5 and SW 2023 SP3.

Deleting the registry entries as @weianweigan proposes seems to fix the issue but I'm deploying an add-in to 10 people or so and it's not very practical.

Are the any update on this?

artem1t, 2023-10-18:
Because command tabs can be assigned with custom names (e.g. user can rename those) it can be unsafe to delete the command tabs automatically as there may be a user-assigned command tab. This is why xCAD does not clear command tabs (it might be worth revisiting this). You can either remove the command tab registry key in the installer itself (as a custom action) or you can use the SW API in the OnConnect in xCAD add-in to remove the old tab if exists:

~~~ cs
public override void OnConnect()
        {
            var legacyTab = this.CommandManager.CmdMgr.GetCommandTab((int)swDocumentTypes_e.swDocPART, "MyLegacyTab");
            if (legacyTab != null) 
            {
                this.CommandManager.CmdMgr.RemoveCommandTab(legacyTab);
            }
~~~