xarial/xcad #90: multiple instances of same control group in 1 tab?

MJyee, 2023-02-20:
i have a group of controls that i would want to have multiple instances of in a single group. but when i try to do it, the show pmp section returns an error " System.Exception: 'Tag is not unique' " how do i fix this ?

additionally, i have a selectionbox in the group that i would like to change its selection color on the definition, how would i do it as well.

> [DisplayName("Values")]
>   [Tab]
>  public class ValuesTab
>    {
>        [Title("Loop 1")]
>        public loopGroupBase loop1 { get; set; }
>
>        [Title("Loop 2")]
>        public loopGroupBase loop2 { get; set; }
>   }

artem1t, 2023-02-20:
In order to specify the color you need to use the

artem1t, 2023-02-20:
Regarding the reusing of the instances. Do you have [ControlTag] or [Metadata] attributes used in the loopGroupBase class?

MJyee, 2023-02-20:
you mean if u have any control tags it will cause that error ?

here is the code for the group,

{
    public enum selboxnumber
    {
        one = StandardSelectionColor_e.Primary,
        two = StandardSelectionColor_e.Secondary,
        three = StandardSelectionColor_e.Tertiary,
    }
    public enum GDTOptions
    {
        position,
        none
    }

    [CheckableGroupBox("ToggleGroupCheckbox")]
    public class loopGroupBase : INotifyPropertyChanged
    {
        //internal using valus 
        [ExcludeControl]
        public selboxnumber _selboxnumber { get; }

        private string DistanceValueString;
        public loopGroupBase()
        {
            ischecked = false;
            //_selboxnumber = Selboxnumber;
            leftButton = onLeftClick;
        }

        private void onLeftClick()
        {
            Distance_val = Guid.NewGuid().ToString();
        }

        public event PropertyChangedEventHandler PropertyChanged;

        //generation infos
        [Metadata("ToggleGroupCheckbox")]
        public bool ischecked { get; set; }

        [Label("Planar faces")]
        [SelectionBoxOptions(Xarial.XCad.UI.PropertyPage.Enums.SelectionBoxStyle_e.MultipleItemSelect)]
        [ControlOptions(height: 30)]
        public List<IXFace> faces { get; set; }

        [TextBoxOptions(TextBoxStyle_e.NoBorder)]
        [ControlOptions(AddControlOptions_e.Visible, backgroundColor: KnownColor.Transparent, textColor: KnownColor.ControlText)]
        public string Distance_val
        {
            get => DistanceValueString;
            set
            {
                DistanceValueString = value;
                this.PropertyChanged?.Invoke(this, new PropertyChangedEventArgs(nameof(Distance_val)));
            }
        }

        [BitmapButton(BitmapButtonLabelType_e.StackLeft)]
        [ControlOptions(left: 30, top: 100)]
        public Action leftButton { get; set; }

        [BitmapButton(BitmapButtonLabelType_e.StackRight)]
        [ControlOptions(left: 60, top: 100)]
        public Action rightButton { get; set; }

        [Label("Tollerance/Precision")]
        //[StandardControlIcon(BitmapLabelType_e.)]
        [ControlTag("State")]
        public TollernceOptions combobox { get; set; }

        [Icon(typeof(Resources), nameof(Resources.Company_logo))]
        public string textbox { get; set; }

        [DependentOn(typeof(visibilityHandler), "State")]
        [Icon(typeof(Resources), nameof(Resources.Company_logo))]
        public string textbox2 { get; set; }

        [Title("GD&T")]
        [ControlTag("GDT")]
        public bool GDT_enable { get; set; }

        [DependentOn(typeof(EnableHandler), "GDT")]
        public GDTOptions GDTOptionsBox { get; set; }

        [DependentOn(typeof(EnableHandler), "GDT")]
        [Icon(typeof(Resources), nameof(Resources.Company_logo))]
        public double GDT_value { get; set; }
    }

    // dependencies Handlers
    public class visibilityHandler : IDependencyHandler
    {
        public void UpdateState(IXApplication app, IControl source, IControl[] dependencies)
        {
            source.Visible = ((TollernceOptions)dependencies?.First().GetValue() != TollernceOptions.Symmetric);
        }
    }

    public class EnableHandler : IDependencyHandler
    {
        public void UpdateState(IXApplication app, IControl source, IControl[] dependencies)
        {
            source.Enabled = (bool)dependencies.First().GetValue();
        }
    }

MJyee, 2023-02-20:
> In order to specify the color you need to use the SelectionColor property in SelectionBoxOptions attribute as shown [here](https://github.com/xarial/xcad-examples/blob/master/PMPage/cs/Page/Groups/AdvancedSelectionBoxGroup.cs#L48-L49). You ill have an option to select primary, secondary colors based on the system settings

for this i know its for static of it but my part is that for each instance of the loop, ill want it to cycle through the 3 selection color options, if its not possible then its fine.

artem1t, 2023-02-26:
Thanks for the clarification. The [Metadata] and [ControlTag] attributes define the global references, e.g. you can reference them from any level at any structure (e.g. group and tab). That means that the class cannot be reused to create another group or tab and it is required to have new class for every group (e.g. copy of loopGroupBase class).

Another option (I will need to check this) is to make the properties like GDT_enable virtual in loopGroupBase and overwrite in the class which inherits from loopGroupBase to change the name of the [ControlTag] value (e.g. GDT1). I will check if this works and get back to you.