xarial/xcad #93: Assign keyboard shortcut for commands

andrMollo, 2023-04-19:
As per title, is it possible to assign a user-defined keyboard shortcut to commands added via xCAD?

artem1t, 2023-04-22:
Yes, it should be possible just as with any other standard command in SW

andrMollo, 2023-04-22:
Thanks for the reply. I thought of writing directly to the registry key as suggested [here](https://www.cadoverflow.com/t/key-board-shortcut-to-custom-soldiworks-addin-command/374/2?u=andrmollo), because I don't think there is an API method to do it. It would be awesome if xCAD could implement it.

artem1t, 2023-04-22:
Ah, sorry, you mean you want to assign this from the code, not by the user, right?

andrMollo, 2023-04-22:
Yes, l'm sorry if it wasn't clear. I'm using your framework to create an addin with some button to lunch VBA macros (like your toolbar+ that I'm not allow to use at my company). If I were able to assign shortcut with the code I could standardize the user experience for all users. I have not tried to assign a shortcut to a button via the SW user interface.

andrMollo, 2023-04-25:
Hello! I was able to add keyboard shortcuts via the SolidWorks user interface, that's awesome. I you prefer I can close the issue.

andrMollo, 2023-12-19:
Hi! Sorry to bring it up again. I was able to assign shortcuts via the SolidWorks user interface but sometimes they get switched.

I can't pin-point why, but the shortcut I assigned to a command of the add-in is then assigned to another one.

Can you help me or show me an area in which I can investigate the issue? Right now I'm clueless.

Thanks!