xarial/xcad #94: can not use npgsql.dll

Brent-bai, 2023-04-30:
my application raise error when connect my postgresql database by npgsql.dll, the vs reports " can not load the System.Runtime.CompilerServices version 4.0.2.0,..."
it seems you need update the depencities.

Brent-bai, 2023-05-01:
I have tried all the method, it does not work. ref url:
https://stackoverflow.com/questions/62764744/could-not-load-file-or-assembly-system-runtime-compilerservices-unsafe

artem1t, 2023-05-01:
Which target framework do you use?

Brent-bai, 2023-05-04:
.net framework 4.7.2. 
I have changed the project to .net Core 6, it worked well.  
Thanks for the reply.
Another question, is there a method to replace the doc's reference in class IXDocument? I can not find it.  I want to modify the component's file name while also modifying the drawing's file name without losing the reference.

Brent-bai, 2023-05-04:
> Which target framework do you use? I have seen similar issues with .NET Core and .NET6/7. Also, check your output folder, do you have this dll available of this specific version? If not most likely you have some references which are referencing different version and as you cannot use binding redirects on add-ins you need to use the AssemblyResolve on the AppDomain explicitly to replace the DLL reference.

.net framework 4.7.2, 4.8.1 also tried.
I have changed the project to .net Core 6, it worked well.
Thanks for the reply.
Another question, is there a method to replace the doc's reference in class IXDocument? I can not find it. I want to modify the component's file name while also modifying the drawing's file name without losing the reference.

Brent-bai, 2023-05-04:
Sorry, I find the IXDocumentDependencies method to replace the component of the drawing, but parameter only can be IXDocument3D document type. When i use the OnModelSaving handler, the new component file not yet saved, so it can not get the IXDocument of the new one, and the replace method can not go ahead. seems  I need using the sw document manager to resolve it.  The parameter of replace method changing to the file path may be useful.

Brent-bai, 2023-05-17:
I have tried AssemblyResolve, it also worked.

closed 2023-05-17 by Brent-bai