xarial/xcad #97: event PropertyValueChangedDelegate ValueChanged not work well

Brent-bai, 2023-05-10:
Event PropertyValueChangedDelegate ValueChanged not work well when I use the solidworks customize properties tab. ValueChanged method does not really calculate the difference between the old and new value. I hope maybe the method store the old value, so I can add a function to calculate the value does chaged or no.

artem1t, 2023-05-10:
Please try the dev build (or I will publish beta soon). I think the issue is the SW event for properties was not operating properly for some time (e.g. only raised when API changes property, not the user), so xCAD has a complex workaround where it checks when the Properties dialog is open and closed and calculates the differences which seem like has a bug in it. It seems like SW has finally fixed the issue with the original event so the dev build reverts back to the actual sw event which should fix this problem. Please let me know what version of SW do you use and I will double check this.

Brent-bai, 2023-05-11:
> 请尝试开发版本（否则我将很快发布测试版）。我认为问题是属性的 SW 事件在一段时间内无法正常运行（例如，仅在 API 更改属性而不是用户时引发），因此 xCAD 有一个复杂的解决方法，它检查属性对话框何时打开和关闭并计算差异似乎有一个错误。似乎 SW 终于解决了原始事件的问题，因此开发版本恢复到应该解决此问题的实际 sw 事件。请让我知道您使用什么版本的软件，我会仔细检查这一点。

hi, Visual studio 2022, solidworks2019, Xcad.solidworks 0.8.0-beta.4627

artem1t, 2023-05-19:
The underlying SW event only started to work properly in SW 2020, I am afraid (you can check with the ProeptyChanged event from the IPartDoc). I checked in SW 2019 and it does not fire, ut fires in the later versions.

closed 2023-07-12 by Brent-bai