xarial/xcad #98: some question about the document event handler

Brent-bai, 2023-05-11:
When I use sw Saveas function, Xcad can not catch the old file closing or destroyed event.
And the Application.Documents.RegisterHandler can not return a value, so I can not know when the DocumentSaveArgs have done his job. 
As a example, I want to delete the old file after changing the name of the a part(in Saveas function), but I can not add any function in the DocumentHandler, because the DocumentSaveArgs wont release the old file before the DocumentHandler has done the whole job. So I add a new eventhandler to catch the Saveas function as releasing the old file, but it does not worked.

artem1t, 2023-05-11:
Interesting question. SOLIDWORKS seems to be not changing the point of the IModelDoc2 (ISwDocument) when the file is saved as (renamed) so the original pointer stays in alive and thus the file is not closed reopened so the Destroyed is not called and RegisterHandler as well.

I think what you can do is within the SaveAs handler subscribe to IXApplication::Idle notification and this should be raised after the operation is complete so you can subscribe from this event and delete your old file. To be more safe you can check if the file released before deleting and only unsubscribe from Idel once it is released.

Brent-bai, 2023-05-12:
Thank you, IXApplication::Idle worked.

closed 2023-05-12 by Brent-bai