xarial/xcad #101: About PMPage problem

Brent-bai, 2023-05-25:
I encountered the following issues when using the PMPage interface: 1 The custom wpf interface only displays a portion in the property manager page; 2. Can a label be added on the same line before the text box to identify the content to be filled in this text box; 3. It seems that the webview2 control cannot be used for wpf in the class library, and the display is a blank page

Brent-bai, 2023-07-10:
How do I chage the size of the user control wpf view (the yellow box is the true size)  which in a pmp page.
[image]

Brent-bai, 2023-07-10:
I chage a method: add the Mgrtab. 
`private void CreatFeatureMgrTab()
        {
            var doc = Application.Documents.Active;
            if (doc == null)
            { return; }
            // prp_m_page.Show(prp_m_model);
            if (tab == null)
            {
                tab = CreateFeatureManagerTab<prpFeatureMgrControl>(doc);
                
                tab.Control.Sw = Application.Sw;
            }
            else
            {
                tab.Close();
                tab.Dispose();
                tab = null;
            }

        }`
It also does not work well, when I click the tab, it only display part view of the tab:
[image]
So i need adjust the size of whole tab.

artem1t, 2023-07-12:
> How do I chage the size of the user control wpf view (the yellow box is the true size) which in a pmp page. [image]

You can use [ControlOptionsAttribute](https://github.com/xarial/xcad/blob/master/src/Base/UI/PropertyPage/Attributes/ControlOptionsAttribute.cs#L27)

Brent-bai, 2023-07-12:
Thanks you very much! It worked.

closed 2023-07-12 by Brent-bai