xarial/xcad #104: Commands are dropped when CommandItemInfo is applied.

geoffgscott, 2023-07-07:
If `CommandItemInfo` is applied the commands disappear from the command tab (for all document types). Tested on both SW2021 and SW2022.

```cs
[Title("Power Tools")]
public enum Commands_e
{
    // below line causes command to disappear
    [CommandItemInfo(WorkspaceTypes_e.AllDocuments)]
    [Icon(typeof(Resources), nameof(Resources.electric_drill))]
    Drill,
    [Icon(typeof(Resources), nameof(Resources.electric_screwdriver))]
    Screwdriver,
    [Icon(typeof(Resources), nameof(Resources.power_saw))]
    Saw
}
```
Minimal reproduction here:
[https://github.com/geoffgscott/CommandPanelTest](https://github.com/geoffgscott/CommandPanelTest)

artem1t, 2023-07-10:
Did you run the installer on the dev machine (e.g. where it was compiled)? If so, please make sure that the dev version is not registered:

•	Clear the solution
•	Start SOLIDWORKS to see if there is no add-in
•	Install from the installer
•	Validated that files in the installation folder match the build (e.g. there are no missing files)

Users reported issues with the VSI installer (COM objects were not registered). I am using WiX

Try to add Try-catch block within your OnConnect to see if there is an exception (it can be an exception if you have previous dev version of the add-in registered due to the conflict of the ids)

If the above does not work, please try to register manually with regasm /codebase.

If none of the above works, we can schedule an online meeting to troubleshoot.

geoffgscott, 2023-07-12:
To clarify this appears to be unrelated to the installer at all. Issue is noticed both in local development (even after running a `project clean`) as well as in a bundled install (installed to a separate clean machine instance).

Looks to be specifically related to this line ` [CommandItemInfo(WorkspaceTypes_e.AllDocuments)]`

artem1t, 2023-07-12:
@geoffgscott we can organize an online meeting to resolve this. I have sent you an e-mail