xarial/xcad #105: FlatPatternActivator

fatih1445, 2023-07-12:
FlatPatternActivator not working on multibody sheet metal parts.

[image]

artem1t, 2023-09-09:
Could you please attach this sample model?

fatih1445, 2023-11-09:
[gl_ku007.zip](https://github.com/xarial/xcad/files/13312261/gl_ku007.zip)

artem1t, 2023-11-09:
Thank you for the sample model. The issue here is that the active configuration of this file is a SM-FLAT-PATTERN and you cannot create flat pattern from the flat pattern configuration. You can try this by trying to create flat pattern of this file manually (there is not flat pattern view).

Just activate your top configuration and it will work. In version 0.8.X there is a better exception for this case

`Xarial.XCad.SolidWorks.Documents.Exceptions.InvalidFlatPatternConfigurationException: 'The flat pattern drawing view is invalid as it does not contain the flat pattern feature in the flattened state. This is usually caused by an invalid SM-FLAT-PATTERN configuration in the part file. Try removing this configuration'
`

fatih1445, 2023-11-10:
Thank you very much for your time.

closed 2023-11-10 by fatih1445