xarial/xcad #107: SolidWorks Addons failing to load in SolidWorks 2024

levihilton, 2023-09-08:
Addons built using this package (v0.8.0-beta.4687) are failing to load in SolidWorks 2024 pre-release. Is this something that happens with each new release of SolidWorks?  Is there something I can do on my side to provide forward compatibility?

artem1t, 2023-09-09:
I will check with Pre Release version. It was working OK with Beta 2024 version for me when I checked it before. In general, you should not do any additional steps to make it compatible. Could you please clarify the target NET framework that you use and the symptoms (does it fail in OnConnect or does OnConnect never reach)?

Blogbotana, 2023-11-08:
I have Solid 2023 and also can't use this library

SubhraMaiti, 2024-08-12:
I used the below command to register the dll and add in is working in SolidWorks 2024. 

%windir%\Microsoft.NET\Framework64\v4.0.30319\RegAsm.exe "address of the dll file" /codebase

foreachthing, 2025-03-25:
@SubhraMaiti, I only get `RegAsm : error RA0000 : Could not load file or assembly 'PresentationFramework, Version=8.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies. The system cannot find the file specified.`
It used to work in 2022 (currently installed SolidWorks 2024 and .net8.0)... now I am stuck with developing... :-(
Any news/ideas?