xarial/xcad #108: About pmp cancel button and dependency controls

Brent-bai, 2023-09-12:
1. As I click the cancel button in a pmp,  I want make a dialog in order to remind the user to confirm the cancel opertation. I found that statement "arg.Cancel = true;" would not work in case "reason == PageCloseReasons_e.Cancel"
2. Can a checkbox control two dependency controls to disable or not. I have tried, it dose not worked. 
Example: 
checkbox A checked, checkbox B enabled , as B checked, the textbox 1 enabled; as B unchecked, the textbox 1 disabled;
checkbox A unchecked, checkbox B disabled, and textbox 1 disabled;

Brent-bai, 2023-11-02:
Found another problem:
Two ControlGroup altered by dependecy options, controled by two icon button; 
The ControlGroup does changed when I click each buttons; but input values of disabled ControlGruop remained, when I change it back, it would cause unexpected result (because it will do something when all the input is difined). It seems just hide but not disabled. Because when I switch to another control group, the input focus remains in the previous control group.
In the CustomControlDependantHandler, I add the "source.Dispose();", it would not work.
Do you have any ideas?
Thank you!

Brent-bai, 2023-11-02:
I have add a reset() method to ControlGroup to reset the remained value or selection to default, but the input focus problem still existed. I noted the IControl resource code have the focus() method, but not finished. Would it be the answer in the future?

Brent-bai, 2023-11-03:
I use reflection to get the IPropertyManagerPage2, but found the control id is changed everytime. 😂

artem1t, 2023-11-06:
Sorry, a bit confused with the last issue with ControlGroup. Do you have code snippet for that?

Brent-bai, 2023-11-06:
[image]
This gif will explain the problem. The CustomControlDependantHandler5 should disable the controlgroup. But it still enabled, just hidden.
`
//DependantHandler
public class CustomControlDependantHandler5 : IDependencyHandler
{
    public void UpdateState(IXApplication app, IControl source, IControl[] dependencies)
    {
        var res = (bool)dependencies.First().GetValue();
        source.Enabled = res;
        source.Visible = res;
	}
}

//ControlGruop
[Title("-FourCorner")]
[DependentOn(typeof(CustomControlDependantHandler5), nameof(isFourCornerPoints))]
public PointsFromFourCornerControlsGroup? pointsFromFourCornerControlsGroup { get; set; }

[Title("-OneCorner")]
[DependentOn(typeof(CustomControlDependantHandler5), nameof(isOneCornerPoints))]
public PointsFromOneCornerControlGroup? pointsFromOneCornerControlsGroup { get; set; }

//Icon Toggle Button
 private bool _isFourCornerPoints;
 [DependentOn(typeof(CustomControlDependantHandler3), nameof(holeReferenceType))]
 [BitmapToggleButton(typeof(Resource1), nameof(Resource1.FourCornerPoint), BitmapEffect_e.None | BitmapEffect_e.Transparent, 32, 32)]        
 [ControlTag(nameof(isFourCornerPoints))]
 [Description("四角点")]
 [ControlOptions(left: 0, top: 0)]
 public bool isFourCornerPoints
 {
     get => _isFourCornerPoints;
     set
     {
         if (_isFourCornerPoints != value)
         {
             _isFourCornerPoints = value;
             PropertyChanged?.Invoke(this, new PropertyChangedEventArgs(nameof(isFourCornerPoints)));
             if (value)
             {
                 isOneCornerPoints = false;                 
             }
             else
             {
                 if (pointsFromFourCornerControlsGroup!=null) pointsFromFourCornerControlsGroup.Reset();                       
             }
         }        
     }
 } 
`

Brent-bai, 2023-11-06:
By the way, may the property "IsEditing" of ISwSketch2D be public? So I can get the actived sketch.

artem1t, 2023-11-08:
Ah, I see, thank you. The issue here is that IPropertyManagerPageGroup does not have an Enabled property in SW API and it is not considered as control (unless I missing this). in xCAD group is also a control so it has this property derived, but implementation is empty: https://github.com/xarial/xcad/blob/dev/src/SolidWorks/UI/PropertyPage/Toolkit/Controls/PropertyManagerPageGroupControl.cs#L44-L46. I think it should throw NotSupported exception