xarial/xcad #110: Limit number of selections allowed in Selections Boxes in PMP

MJyee, 2023-10-05:
hi, is there a way to limit the number of selections in a selection box to an example 2 ?
i tried using the custom selection filters to cast the selbox to propertymanagerpageselectionbox to check its count property but it returns a fail cast.

```C#
public class Max2FacesSelection : ISelectionCustomFilter
    {
        public void Filter(IControl selBox, IXSelObject selection, SelectionCustomFilterArguments args)
        {
            int sel_count = ;
            
            if (sel_count > 2)
                args.Reason = "Max 2 Selections Allowed";
            else
                args.Filter = true;
        }

    }
```
my selection box is define like this

```c#
 [SelectionBoxOptions(SelectionColor = StandardSelectionColor_e.Primary, Style = SelectionBoxStyle_e.MultipleItemSelect, CustomFilter = typeof(Max2FacesSelection),)]
        [ControlOptions(height: 50)]
        [Label("Selection")]
        public List<IXFace> Selections { get; set; }
```

artem1t, 2023-10-05:
You were very close. Please see below. I use **?** as value can be null initially

```
public class Max2FacesSelection : ISelectionCustomFilter
    {
        public void Filter(IControl selBox, IXSelObject selection, SelectionCustomFilterArguments args)
        {
            int? sel_count =((System.Collections.IList)selBox.GetValue())?.Count;
            
            if (sel_count > 2)
                args.Reason = "Max 2 Selections Allowed";
            else
                args.Filter = true;
        }
    }
```

MJyee, 2023-10-05:
thanks for the help. that solved the initial issue but now when i try running it the count remains 0 even though i have faces selected in the selection box.
[image]

is it due to some failed cast ?

artem1t, 2023-10-06:
Strange, please make sure this is a correct selection box (I can see you have more than one, perhaps it is just a filter for another box)? It works OK for my test (with beta version of xCAD). Note, that it shows the previous value, e.g. the current filtering object is not yet in the list (e.g. when you select first object, the sel_count will be null or 0, second 1, etc.)

MJyee, 2023-10-06:
Ok after updating to Pre-Release 4627 it is now working, i was originally using the stable 0.7.12

closed 2023-10-06 by MJyee