xarial/xcad #112: Create Assembly Feature result in Exception: Preview context is not specified

MJyee, 2023-10-09:
hi Im trying to make my feature be an assembly feature, the feature itself is just a measurement feature so it does not need to show any preview or generate any bodies, how do it fix this exception ?

artem1t, 2023-10-10:
You can either create a base macro feature without the property page (you will need to manage it yourself, e.g. inherit SwMacroFeatureDefinition<TParams>) or you can provide the preview context by overriding the ProvidePreviewContext method. But you are right, that preview context should not be called if there are no bodies, so I have changed this in the new version (please upgrade to the latest beta) so it will work without context unless you have preview bodies (in this case use the override method mentioned above to specify the context (e.g. component, by default it is using the editing component, but you can assign any component)

MJyee, 2023-10-10:
Thanks for your help my macro now works in assemblies

closed 2023-10-10 by MJyee