xarial/xcad #114: About ISw Type questions

Brent-bai, 2023-10-12:
I noticed that the item of ISwBody.Faces is IXFace type, as I convert it to ISwFace using "as",  the method Entity.GetDistance() will not obtain the value of two face. I used iModelDoc2.ClosestDistance()  instead to resolve this problem.
And there is a similar issue. When I added the specified PlanarFaces to the List<ISwPlanarFace> and when iterating, List.Contains() did not return the correct judgment, resulting in duplicate objects in the list. I used Tag to resove it.
I can not find the reason of it. As a novice in programming, it is possible that I am not familiar with C#, and I hope someone can answer my questions. Thank you.
sw:2021
xcad: 0.8.0 beta 4839

Brent-bai, 2023-10-12:
I have tried to set the List type to List<IFace2>, the Contain() method return the right bool value, but List<ISwface> not.

Brent-bai, 2023-10-16:
Another problem: IXPart.Bodys[0].Faces return IEnumerable<IXFace> , which contains SwPlanarFaces and SwCylindriacalFaces, and all dispath or Entity is Face2Class. ISwPart.Bodys[0].Faces return same IEnumerable<IXFace>, but there are system com_object entities in the list, casuing some errors, as shown in the below picture.  And it's not happened everytime by same part.
[image]

artem1t, 2023-10-18:
I think this might be related to threading/COM with those dispatches. Dispatch is what SW provides. To compare the entities I recommend using the IXObject::Equals which will compare dispatches using the ISldWorks::IsSame. This should return true for the same entities even if one of those is __ComObject and other is IFace2

Brent-bai, 2023-10-19:
> I think this might be related to threading/COM with those dispatches. Dispatch is what SW provides. To compare the entities I recommend using the IXObject::Equals which will compare dispatches using [...]

IsSame worked! Thank you very much.

closed 2023-12-05 by Brent-bai