xarial/xcad #116: A sketchpoint issue

Brent-bai, 2023-11-07:
```
Point point = new Point(X,Y,0.0);
swSketchPoint = (ISwSketchPoint)sketch2D.Entities.PreCreatePoint();
swSketchPoint.Coordinate = point;
sketch2D.Entities.Add(swSketchPoint);
```
This code cause sw app crashed with no reason.
I change the code to solidworks api as below:
```
SketchManager swSketchMgr = (SketchManager)swModel.SketchManager;
Sketch swSketch = swSketchMgr.ActiveSketch;
swSketchMgr.CreatePoint(x, y, 0.0);
swModel.ClearSelection2(true);
swModel.GraphicsRedraw();
```
It also cause the app crashing. The CreatPoint() throw a exception:  "This is often an indication that other memory is corrupt.”
I ask the ChatGpt, it changed the code to:
```
SketchManager swSketchMgr = (SketchManager)swModel.SketchManager;
Sketch swSketch = swSketchMgr.ActiveSketch;
swModel.ClearSelection2(true);
swModel.SetAddToDB(true);
swSketchMgr.CreatePoint(x, y, 0.0);
swModel.SetAddToDB(false);
swModel.GraphicsRedraw();
```
I noticed the code adding a SetAddToDB() method, I wonder that xcad.net api having considered this issue.

Brent-bai, 2023-11-08:
I create a minium environment, only click a command to run above function, they all work well. In my code, first creat a sketch, then click a command to start a poperty manager page, as I click one button of the page to run the first code above, the sw app crashed. It seemes sw crashing when returning to my command page.
Then I add the AddHorizontalDimension2() to the third code above, the sw app crashed again.  It is highly suspected that a memory access issue occurred when the sketching tool's property manager switched back to my property manager page.

Brent-bai, 2023-11-08:
I'm sure it does cause by my pmp closing when sketchtool raising. This command may not use pmp.

Brent-bai, 2023-11-08:
There is a constance "OPTS_DEFAULT" in internal Show() which in  Internal class PropertyManagerPagePage of Xarial.XCad.SolidWorks.UI.PropertyPage.Toolkit.Controls, it allow the other page to be a StackPage. Make it to a page attribute may resolve this issue.

artem1t, 2023-11-08:
Yes, that makes sense to add an option to have StackPage. I will add one.

Regarding the sketch. Yes, AddToDB is used: https://github.com/xarial/xcad/blob/!dev/src/SolidWorks/Sketch/SwSketchEntityCollection.cs#L73C2-L89

But if I am not mistaken this was added into the pre-release version (e.g. 8.X and not available in 7.X)

Brent-bai, 2023-11-22:
I have changed "OPTS_DEFAULT = 1", it does not worked. The property manager page will not being closed when I click the OK or Cancle button.  Here is my sample code:
[SketchCreateTest.zip](https://github.com/xarial/xcad/files/13433318/SketchCreateTest.zip)

Brent-bai, 2023-11-27:
I used  lockedPage attribue ([PageOptions(PageOptions_e.LockedPage)]) option to resolve the problem, avoiding the crashing of sw when inserting a point and adding dimension. But unfortunately in this option, the SelectById2 method will not work. Because some sketch points are com_object, it can not use Select4 method to select the point. It seems I have tried all the way, but I can't find the good answer.