xarial/xcad #128: Development of .NET 6.0/7.0 add-in without close SolidWorks

jesse-21499, 2024-06-08:
Is it possible to modify source code of .NET 6.0 or 7.0 SW add-in (allowing build and write to dll after stop debugging and unload add-in in SW) without close and restart SolidWorks?
When I try to do this, dll file is locked for writing and I need to exit and restart solidworks.
I'm using Xarial.XCad.SolidWorks (0.8.0) and .NET 7.0

artem1t, 2024-06-16:
I am afraid it won't be possible. .NET dlls are loaded into the process so it cannot be unloaded from the process. There is a way to load into a different app domain, but I would not recommend going this route. But you can edit the code (most of it) ad-hoc while debugging via hot reload.

jesse-21499, 2024-06-17:
Thanks for the information.
I've tried using XArial to develop add-in that access methods from 3'rd party add-in (C++ based) loaded in SW.
External C# exe (console,winform or WPF) is good for temporarily develop and debug purposes without restart SW but is not possible to pass the IActiveDoc document argument to C++ based add-in methods. Addin is needed.
So I will build portions of code that access 3'rd party C++ API inside auxiliary add-in and try to use HotReload.
Also tried using C++ COM NonAttributed Addd-in from SW API but after unload add-in, SW takes about 15 or 20 minutes to free or release COM pending objects,call ExitInstance() and unload DLL,leaving it unlocked for writing.

Sorry for my english
Regards

spiriapbergeron, 2024-09-08:
I can use "edit-and-continue" in my addin most of the time no problem. I have never tried to modify xcad itself though