xarial/xcad #130: How to Register Icons for Customization?

foreachthing, 2024-07-01:
If I want to customize a certain toolbar, I cannot access the icons (commands) from my addin (swTools4):
[image]
No icons (commands) listed. Even though my toolbar (and menu) are full with commands.

Please compare to "Surface":
[image]
All possible commands are available.

Any help is appreciated.

andrMollo, 2024-07-03:
Hi there,
I reached out to @artem1t some time ago: it seems there is a bug in SolidWorks, the icons are actually there they are just tiny. You can try to mouse over and see if the tooltip is displayed.

I hope it helps.

foreachthing, 2024-07-04:
@andrMollo, yes, it does help - but not. 😦 
I don't undestand why this is a SolidWorks-Bug ... since when I made my toolbar without xcad, it worked as expected. Any example project I've found, using xcad, does not show the icons. So, I'll leave this issue open as it only happens when using xcad.

artem1t, 2024-07-04:
@foreachthing, you are correct. I have just done a test and it is indeed displayed correctly with vanilla SOLIDWORKS API, @andrMollo, I apologize as I was mistaken. xCAD is still using the vanilla API (e.g. calling the same Command Manager API) but something breaks this particular icon rendering, perhaps image encoding or something like that. I will be working on this now as I have a working example with SW API so can compare where it is breaking.

foreachthing, 2024-07-05:
Sweet!!!
[image]

Thanks for your hard work!! 😀

artem1t, 2024-07-05:
Fixed in dev build. The issue was related to disposing of temp icons. Icons were removed immediately after the toolbar is created which has caused this issue. Now the icons are removed when the group is removed.

closed 2024-07-05 by artem1t

foreachthing, 2024-07-05:
Yeah, thanks! I saw that and tried to compile it.
Still trying to find out how to compile without all the examples :roll_eyes:

andrMollo, 2024-07-06:
Yesss! Thanks a lot!

andrMollo, 2024-07-17:
Hi @artem1t , I'm using `xCAD` as a NuGet package (`v0.8`). Any chance that this fix will be released there as well?

commit ef9d6bbe (2024-07-05): Fixed #130 - How to Register Icons for Customization?

Added IXAnnotation::Owner
Changed the IXSheet sheet format structure