xarial/xcad #133: Closing SW Popup window doesn't dispose it properly

spiriapbergeron, 2024-09-07:
There's a small logic inversion when closing a SwPopupWindow. This is causing duplicate windows being open when you want to reopen Popup "modeless" windows, when you try to close them yourself by calling Close() in the SwPopupWindow as opposed from pressing the [X] icon.

Also I guess the problem doesn't appear if it's a modal dialogbox because it must get closed.

But since I have a floating windows it was causing big problems.

I see this bug in both Master and Dev (I caught it in Dev).

https://github.com/xarial/xcad/blob/dev/src/SolidWorks/UI/SwPopupWindow.cs

Should be **if (!m_IsDisposed)**, or, more clearly,  **if (m_IsDisposed == false)**

        public override void Close()
        {
            **if (m_IsDisposed)** <----- ouch
            {
                m_IsDisposed = true;
                m_Form.Close();
            }
        }

andrMollo, 2024-10-25:
Hi! I see this is fixed in the dev branch. The same is true for this #130. Any chance we can get it to main and then NuGet?

I tried to use the preview NuGet feed as explained [here](https://xcad.xarial.com/installation/preview-builds/), but I can't get it to work. Is this still available?

Thanks!!

artem1t, 2025-04-09:
Alpha version is now available on NuGet. Dev nuget feed is moved to https://pkgs.dev.azure.com/xarial/xcad/_packaging/xcad-dev/nuget/v3/index.json. Same instructions as in the article, just different url