xarial/xcad #135: Event can not be released

Brent-bai, 2024-09-10:
It's a addin application for solidworks, which contain a wpf window with the viewmodel . 
In the viewmodel, I add two event by using below code:
```
application.Documents.Active.Selections.NewSelection += Selections_NewSelection;
application.Documents.Active.Selections.ClearSelection += Selections_ClearSelection;
```
And another function which getDoc method in the viewmodel, I use the linq statements to get a doc:
```
var doc = application.Documents.First(x => x.Title.Contains(name)); 
``` 
In the viewmodel.dispose() method, it will release the event:
```
application.Documents.Active.Selections.NewSelection -= Selections_NewSelection;
application.Documents.Active.Selections.ClearSelection -= Selections_ClearSelection;
```
The viewmodel.dispose() method would call by wpf window_Closed method.
The issue is:
I used the getDoc method, then closed the window, the events never been released. It always be called until I closed the solidworks application.

artem1t, 2025-04-09:
When you call ::Active document it returns new object (although the actual IModelDoc2 is the same), but IXDocument is different. One of the older version of xCAD was based on keeping track of IXDocument so it never creates new instances, but this was quite a problematic solution so I reverted it, and instead it is now can be compared via Equals to check if documents are the same. For the events management, you just need to store the pointer to the IXDocument, subscribe to event and release from the same pointer, e.g.:

~~~ cs
private IXDocument m_Doc;
...
m_Doc = application.Documents.Active;
...
m_Doc.Selections.ClearSelection +=
...
m_Doc.Selections.ClearSelection -=
~~~

Brent-bai, 2025-04-14:
OK, Thanks!

closed 2025-04-14 by Brent-bai